OpenNuvoton/OpenOCD-Nuvoton #1: Compile Error

damsponz, 2019-02-07:
Hello,

I try to compile OpenOCD-Nuvoton for use with NuMaker-PFM-M2351 because the https://gnu-mcu-eclipse.github.io/openocd/install/ doesn't work with this board.

After install with homebrew installer on MacOS Mojave
- make : GNU Make 3.81
- libtool : libtool-2.4.6_1.mojave.bottle.tar.gz
- pkg-config : pkg-config-0.29.2.mojave.bottle.tar.gz
- autoconf : autoconf-2.69.mojave.bottle.4.tar.gz
- automake : automake-1.16.1_1.mojave.bottle.tar.gz
- texinfo : texinfo-6.5.mojave.bottle.tar.gz

I run this command :
./boostrap : Ok
./configure : Ok
make : NOK

But with the "make" command I have this error :

I /bin/sh ../libtool  --tag=CC --silent  --mode=compile gcc -DHAVE_CONFIG_H -I. -I..    -Wall -Wextra -Werror -fvisibility=hidden  -g -O2 -MT libjaylink_la-discovery_tcp.lo -MD -MP -MF .deps/libjaylink_la-discovery_tcp.Tpo -c -o libjaylink_la-discovery_tcp.lo `test -f 'discovery_tcp.c' || echo './'`discovery_tcp.c
discovery_tcp.c:305:9: error: implicit declaration of function 'select' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
                ret = select(sock + 1, &rfds, NULL, NULL, &timeout);
                      ^
1 error generated.
make[4]: *** [libjaylink_la-discovery_tcp.lo] Error 1
make[3]: *** [all-recursive] Error 1
make[2]: *** [all] Error 2
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

Could you help me ?

Thank you.
Best Regards,
damsponz.

ZaleYu, 2019-02-08:
Hi damsponz:
Do you want to run OpenOCD on Mac?
If not, you can download NuEclipse on the Nuvoton website.
The tool contains OpenOCD which supports M2351.

To compile OpenOCD, you must close the compile warning.
To do that, you can read the readme document inside the OpenOCD folder.
The document teaches you how to close it and how to compile OpenOCD.

Sincerely,
Zale Yu

damsponz, 2019-02-11:
Hi Zale Yu,

Thank you for your answer.
Yes I would like run OpenOCD modify by Nuvoton on Mac.

I read README and README.OSX files but the installation instructions in README.OSX with brew install doesn't allow M2351 board.

When I use manual installation instruction and close compile warning I have some error.
./boostrap : OK
./configure --disable-werror --disable-wextra --disable-gccwarnings : OK
make : NOK

But I have some errors : 

/bin/sh ../libtool  --tag=CC --silent  --mode=link gcc -Wall -Wextra -Werror -fvisibility=hidden -I/usr/local/Cellar/libusb/1.0.22/include/libusb-1.0 -g -O2  -no-undefined  -o libjaylink.la  libjaylink_la-buffer.lo libjaylink_la-core.lo libjaylink_la-device.lo libjaylink_la-discovery.lo libjaylink_la-discovery_tcp.lo libjaylink_la-emucom.lo libjaylink_la-error.lo libjaylink_la-fileio.lo libjaylink_la-jtag.lo libjaylink_la-list.lo libjaylink_la-log.lo libjaylink_la-socket.lo libjaylink_la-strutil.lo libjaylink_la-swd.lo libjaylink_la-swo.lo libjaylink_la-target.lo libjaylink_la-transport.lo libjaylink_la-transport_tcp.lo libjaylink_la-util.lo libjaylink_la-version.lo libjaylink_la-discovery_usb.lo libjaylink_la-transport_usb.lo  -L/usr/local/Cellar/libusb/1.0.22/lib -lusb-1.0 
Makefile:5422: warning: overriding commands for target `check-recursive'
Makefile:4238: warning: ignoring old commands for target `check-recursive'
restore=: && backupdir=".am$$" && \
	am__cwd=`pwd` && CDPATH="${ZSH_VERSION+.}:" && cd . && \
	rm -rf $backupdir && mkdir $backupdir && \
	if (makeinfo --version) >/dev/null 2>&1; then \
	  for f in doc/openocd.info doc/openocd.info-[0-9] doc/openocd.info-[0-9][0-9] doc/openocd.i[0-9] doc/openocd.i[0-9][0-9]; do \
	    if test -f $f; then mv $f $backupdir; restore=mv; else :; fi; \
	  done; \
	else :; fi && \
	cd "$am__cwd"; \
	if makeinfo   -I doc -I ./doc \
	 -o doc/openocd.info ./doc/openocd.texi; \
	then \
	  rc=0; \
	  CDPATH="${ZSH_VERSION+.}:" && cd .; \
	else \
	  rc=$?; \
	  CDPATH="${ZSH_VERSION+.}:" && cd . && \
	  $restore $backupdir/* `echo "./doc/openocd.info" | sed 's|[^/]*$||'`; \
	fi; \
	rm -rf $backupdir; exit $rc

Thank You,
Best regards,
damsponz.

ZaleYu, 2019-02-12:
Hi damsponz:
Have you successfully compiled the official OpenOCD (https://github.com/ntfreak/openocd) on Mac?
If not, please try to do it first.

Sincerely, 
Zale Yu

damsponz, 2019-02-12:
Hi Zale Yu,

I have same error with the official OpenOCD on Mac.

Regards,
damsponz.

ZaleYu, 2019-02-12:
Hi damsponz:
Maybe you can try the following command:
./configure --enable-maintainer-mode 

According to the Readme (https://github.com/openrisc/openOCD/blob/master/README), 
--enable-maintainer-mode  enable make rules and dependencies not useful
                          (and sometimes confusing) to the casual installer
			  NOTE: This option is *required* for GIT builds!
			  It should *not* be used to build a release.

Sincerely, 
Zale Yu

damsponz, 2019-02-14:
Hi Zale Yu,

Sorry, I have same error with this options.

Regards,
damsponz.

ZaleYu, 2019-02-15:
Hi damsponz:
Since I do not have any Mac environment, I am afraid that I cannot help you further.
Maybe you can post your question on the official OpenOCD mailing lists (http://openocd.org/discussion/mailing-lists/). 
From there, you can get support from the OpenOCD experts. 

Sincerely, 
Zale Yu

damsponz, 2019-02-17:
Hi,

Thank you. I will try with this mailing list.

Regards,
damsponz.

filamoon, 2019-05-23:
I ran into the same error and solved it by add one line to src/jtag/drivers/libjaylink/libjaylink/discovery_tcp.c

#include <sys/select.h>