OpenNuvoton/OpenOCD-Nuvoton #5: Failing on OSX with NuMaker-PFM-M2351

suculent, 2019-08-22:
```
Open On-Chip Debugger 0.10.0-dev-ge250e48c-dirty (2019-08-22-17:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : NULINK is Nu-Link1
Error: claim interface failed(-3)
in procedure 'init' 
in procedure 'ocd_bouncer'
```

I've tried to unload default OSX driver (acually was never loaded) and reload using this tutorial:
https://medium.com/@kondal.kolipaka/error-libusb-claim-interface-failed-with-libusb-error-access-c85708c5f43

```
$ ioreg -irc IOUSBDevice
+-o Nu-Link@14200000  <class IORegistryEntry:IOService:IOUSBNub:IOUSBDevice, id 0x100007942, registered, matched, active, busy 0 (68 ms), retain 19>
  | {
  |   "sessionID" = 13806200530033
  |   "iManufacturer" = 1
  |   "bNumConfigurations" = 1
  |   "idProduct" = 20764
  |   "bcdDevice" = 256
  |   "Bus Power Available" = 250
  |   "USB Address" = 20
  |   "bMaxPacketSize0" = 64
  |   "iProduct" = 2
  |   "iSerialNumber" = 0
  |   "bDeviceClass" = 0
  |   "Built-In" = No
  |   "locationID" = 337641472
  |   "bDeviceSubClass" = 0
  |   "bcdUSB" = 272
  |   "USB Product Name" = "Nu-Link"
  |   "PortNum" = 2
  |   "non-removable" = "no"
  |   "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  |   "bDeviceProtocol" = 0
  |   "IOUserClientClass" = "IOUSBDeviceUserClientV2"
  |   "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=3,"CapabilityFlags"=65536,"MaxPowerState"=4,"DriverPowerState"=3}
  |   "kUSBCurrentConfiguration" = 1
  |   "Device Speed" = 1
  |   "USB Vendor Name" = "Nuvoton"
  |   "idVendor" = 1046
  |   "IOGeneralInterest" = "IOCommand is not serializable"
  |   "IOClassNameOverride" = "IOUSBDevice"
  | }
  | 
  +-o Android File Tra  <class IORegistryEntry:IOService:IOUserClient:IOUSBUserClientLegacy:IOUSBDeviceUserClient:IOUSBDeviceUserClientV2, id 0x100007947, !registered, !matched, active, busy 0, retai$
  +-o AppleUSBInterface@0  <class IORegistryEntry:IOService:IOUSBNub:IOUSBInterface, id 0x100007951, registered, matched, active, busy 0 (35 ms), retain 5>
  +-o Android File Tra  <class IORegistryEntry:IOService:IOUserClient:IOUSBUserClientLegacy:IOUSBDeviceUserClient:IOUSBDeviceUserClientV2, id 0x100007952, !registered, !matched, active, busy 0, retai$
  +-o AppleUSBInterface@1  <class IORegistryEntry:IOService:IOUSBNub:IOUSBInterface, id 0x100007954, registered, matched, active, busy 0 (31 ms), retain 5>
```

ZaleYu, 2019-08-29:
Hi suculent:
We did not maintain OSX yet because of the limited resource.

Sincerely, 
Zale Yu

suculent, 2019-08-29:
Well, I am a resource. If you give me some guidance. 

Cheers. 

Odesláno z iPhonu

danielRep, 2021-04-29:
Hi @suculent , were you able to solve this issue?

Ed-Yang, 2022-03-19:
Tested with NuTiny-EVB-NANA103 v2.0 with Nu-Link-Me V3.0.

Modify unlink_usb.c 

#define NULINK_INTERFACE_NUM  2 // 0

It could claim the the interface, but still failed with:

nulink_usb_version failed with cmdSize(4 * 5)
nulink_usb_version failed with cmdSize(4 * 6)

It seems still have problem to communicate with Nu-Link-Me, some help, please.  Thanks!

Ed-Yang, 2022-03-19:
Where could I find the spec. of protocol between OpenOCD and Nu-Link-Me V.3 ?