OpenNuvoton/OpenOCD-Nuvoton #10: No dev-nr in the build output

kristofmulier, 2020-09-08:
<h2>1. Problem explained</h2>

To build OpenOCD from sources, I use the build scripts from Liviu Ionescu (see xPack project at https://xpack.github.io/openocd/). Normally, each OpenOCD I build this way gives a `--version` output like this:

    $ openocd --version
    Open On-Chip Debugger 0.10.0+dev-01404-g393448342 (2020-09-08-10:40)
      Licensed under GNU GPL v2
      For bug reports, read
          http://openocd.org/doc/doxygen/bugs.html

Ignoring the number after `g`, the dev number of the OpenOCD from the example above is `01404`.

However, the Nuvoton OpenOCD gives this `--version` output:

    $ openocd --version
    Open On-Chip Debugger 0.10.0-dev-gb251c4cf (2020-09-08-12:17)
      Licensed under GNU GPL v2
      For bug reports, read
          http://openocd.org/doc/doxygen/bugs.html

As you can see, there is no dev number. There is a series of characters after the `g`, but I don't think that's a dev number.

<h2>2. How I build OpenOCD</h2>

I'll explain how exactly I build OpenOCD. First I install an Ubuntu virtual machine on my computer in Oracle VirtualBox. My current virtual machine is `Ubuntu 20.04.1 LTS`. Then I follow these steps:

<h3>2.1 Install docker</h3>

I issue these commands to install docker:

    $ sudo apt-get clean
    $ sudo apt-get update
    $ sudo apt-get install -y apt-transport-https ca-certificates curl software-properties-common
    $ curl -fsSL https://download.docker.com/linux/ubuntu/gpg | sudo apt-key add -
    $ version_name=$(lsb_release -cs)
    $ sudo add-apt-repository  "deb [arch=amd64] https://download.docker.com/linux/ubuntu  ${version_name} stable"
    $ sudo apt-get update
    $ sudo apt-get -y install docker-ce

To check if docker is properly installed, I run: 

    $ sudo docker run hello-world

To configure docker as regular user:

    $ sudo groupadd docker
    $ sudo gpasswd -a ${USER} docker
    $ sudo service docker restart
    $ shutdown -r now

The last command reboots Linux. Now Docker should work without sudo command. To test, run: 

    $ docker run hello-world

<h3>2.2 Download build scripts</h3>

First I cleanup anything left from a previous build:

    $ sudo rm -rf ~/Downloads/openocd-xpack.git
    $ sudo rm -rf ~/Work

Now I download the build scripts from Liviu Ionescu:

    $ git clone --recurse-submodules https://github.com/xpack-dev-tools/openocd-xpack.git ~/Downloads/openocd-xpack.git

His build scripts need a slight edit, such that the Nuvoton fork is pulled instead of the regular OpenOCD:

    $ cd ~/Downloads/openocd-xpack.git/scripts/
    $ gedit defs-source.sh

The last three lines in `defs-source.sh` should be uncommented and changed into:

    OPENOCD_GIT_URL=https://github.com/OpenNuvoton/OpenOCD-Nuvoton.git
    OPENOCD_GIT_BRANCH=master
    OPENOCD_GIT_COMMIT=HEAD

The build scripts are ready now.

<h3>2.3 Start build</h3>

To do the build, just issue the command:

    $ bash ~/Downloads/openocd-xpack.git/scripts/build.sh --all

Eventually you will find the build output in `~/Work/openocd-0.10.0-14/deploy`. The beauty of this procedure is that it generates builds for Windows 32b, Windows 64b, Linux 32b and Linux 64b!

<h2>3. Question</h2>

Can anyone familiar with this Nuvoton OpenOCD fork try the build procedure described above and see how to fix the dev-nr? We need a fix for the development of our microcontroller IDE "Embeetle". Embeetle IDE parses the OpenOCD `--version` output to identify the exact version, before using it. If the `--version` output doesn't provide a dev-nr, our IDE cannot identify it properly.

ZaleYu, 2020-09-10:
Hi Mulier:
Sorry for the delay of response. 
We built OpenNuvoton OpenOCD in two ways:
1. On Windows, we use MSYS2 to build the win32 version. The OpenNuvoton OpenOCD of Windows version can display the dev number, as you wished.
Open On-Chip Debugger 0.10.0-**dev-00408**-g275b54d5-dirty (2020-08-13-11:52)

2. On Linux, we use Liviu Ionescu's build script to build the Linux version.

Can Embeetle identify the dev-00408?

kristofmulier, 2020-09-10:
Hi Mr. Yu,
Thank you very much for your reply.
Yes, Embeetle can perfectly identify the dev number in a string like:

Open On-Chip Debugger 0.10.0-**dev-00408**-g275b54d5-dirty (2020-08-13-11:52)

I wonder why Liviu Ionescu's build scripts result in no dev number at all? I just test on both Windows and Linux, and get the following output:

    $ openocd --version
    Open On-Chip Debugger 0.10.0-dev-gb251c4cf (2020-09-08-12:17)
    Licensed under GNU GPL v2
    For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html

Could there be a bug in the OpenOCD makefile, such that the dev number gets lost? Or something wrong in the build scripts from Liviu Ionescu?

Note: you can see the complete build procedure, stipulated in great detail, in my original question.

ZaleYu, 2020-09-11:
Hi Mulier:
We have made sure the RELSTR is the root cause.
In the src/Makefile.am:
# banner output includes RELSTR appended to $VERSION from the configure script
# guess-rev.sh returns either a repository version ID or "-snapshot"
if RELEASE
libopenocd_la_CPPFLAGS += -DRELSTR=\"\"
libopenocd_la_CPPFLAGS += -DGITVERSION=\"\"
else
libopenocd_la_CPPFLAGS += -DRELSTR=\"`$(top_srcdir)/guess-rev.sh $(top_srcdir)`\"
libopenocd_la_CPPFLAGS += -DGITVERSION=\"`cd $(top_srcdir) && **git describe**`\"
endif

dev-**00408** derived from the command of **git describe**.
On my computer, I can make the command and get the result:
$ git describe
v0.9.0-**408**-g275b54d5

Therefore, please make sure your **git describe** works properly.

Sincerely,
Zale Yu

kristofmulier, 2020-09-11:
Hi Mr. Yu,
Thank you very much for your reply.

I just tried the command `git describe`. First I clone the OpenOCD Nuvoton repo:

    $ git clone https://github.com/OpenNuvoton/OpenOCD-Nuvoton.git --recurse-submodules
    Cloning into 'OpenOCD-Nuvoton'...
    [...]
    $ cd OpenOCD-Nuvoton/

Then I issue the `git describe` command:

    $ git describe
    fatal: No names found, cannot describe anything.

Apparently this failed. So I tried a few more things:

    $ git describe --all
    heads/master
    $ git describe --tags
    fatal: No names found, cannot describe anything.

What should I do?
Thank you very much for your help.

Kind regards,
Kristof Mulier

ZaleYu, 2020-09-14:
Hi Mulier:
It seemed that the tags of OpenOCD-OpenNuvoton was missing. I have pushed them into OpenOCD-OpenNuvoton. Please pull OpenOCD-OpenNuvoton and try again. Thank you.

Sincerely,
Zale Yu

ilg-ul, 2020-09-14:
> On Windows, we use MSYS2 to build the win32 version. ... On Linux, we use Liviu Ionescu's build script to build the Linux version.

Apart form showing the dev number, what other advantages has running the build on MSYS2 instead of using mingw-w64 on Linux?

ZaleYu, 2020-09-14:
Hi  Liviu Ionescu:
We always use Windows to develop our tools, so MSYS2 is our first choice.

Sincerely,
Zale Yu

ilg-ul, 2020-09-14:
What compiler are you using?

ZaleYu, 2020-09-14:
Build System...x86_64-pc-mingw32

ilg-ul, 2020-09-14:
> x86_64-pc-mingw32

And you think this is better than the mingw-w64 variant? From a pure _nativeness_ point of view, they are both cross-compilers, and I'm trying to understand what are the advantages of running them on MSYS2 vs on Linux.

I'm not trying to sell my scripts, I just don't understand why when there is a common script that in a single invocation can build all 4 images (linux/windows, 32/64), someone would prefer to write another scripts to do a similar thing, but on MSYS2.

I would even bet that there are small inconsistencies between the content of your linux and windows images, for example caused by different library versions.

The xPack build scripts were intended exactly to provide a common and consistent build for all platforms.

But of course, you are free to use any solution you like, I'm just trying to understand use cases, and possibly improve the scripts.

kristofmulier, 2020-09-14:
Hi Mr. @ZaleYu and Mr. @ilg-ul ,
Thank you very much for your help. I've pulled the `master` branch again, and this time the dev number is there:

```sh
$ ./openocd --version
Open On-Chip Debugger 0.10.0-dev-00459-gb251c4cf (2020-09-14-10:20)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
```

I have used the build scripts from @ilg-ul (on my Ubuntu Virtual Machine) as described in my original question, and I can confirm the Nuvoton OpenOCD executable works on both Linux and Windows 10.

Kind regards,
Kristof Mulier

ilg-ul, 2020-09-14:
> I've pulled the master branch again

Did you use any additional git commands that the script is not issuing?

I have difficulties to understand how the builds script works for the linux binaries, and fails to identify the dev-nr for the windows binaries.

kristofmulier, 2020-09-14:
Hi @ilg-ul ,

> "I have difficulties to understand how the builds script works for the linux binaries, and fails to identify the dev-nr for the windows binaries."

This wasn't the case. My original problem was present on both Linux and Windows. I got the following output:

```sh
$ openocd --version
Open On-Chip Debugger 0.10.0-dev-gb251c4cf (2020-09-08-12:17)
  Licensed under GNU GPL v2
  For bug reports, read
      http://openocd.org/doc/doxygen/bugs.html
```

But Mr. @ZaleYu has pushed a fix today. I deleted everything and did the build procedure again from scratch - to make sure that I pull the latest `master` branch during the build, with the fix from Mr. @ZaleYu . Now the `--version` output is good on both Linux and Windows:

```sh
$ ./openocd --version
Open On-Chip Debugger 0.10.0-dev-00459-gb251c4cf (2020-09-14-10:20)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
```

Sorry for the confusion.

Kind regards,
Kristof Mulier

closed 2020-09-15 by ZaleYu