OpenNuvoton/OpenOCD-Nuvoton #11: Failing to build on Linux

vincentdupaquis, 2020-09-16:
Hello,

I try to build on linux, and when launching the ./configure script, he complains very quickly that install.sh and few other scripts are missing. I tried to dig into the doc, but could not find anything looking like what he is searching for ...

What did I do wrong, and how to have it working on Linux ? (I'm running Ubuntu 20.04 at the moment).

The output I obtain : 

**~/tmp/OpenOCD-Nuvoton$ ./configure --enable-parport --enable-ft2232-libftdi --enable-amtjtagaccel
checking for makeinfo... makeinfo
configure: error: cannot find install-sh, install.sh, or shtool in "." "./.." "./../.."**

ZaleYu, 2020-09-17:
Hi:
Please refer to the following link to learn how to build openocd on Linux.
https://github.com/OpenNuvoton/OpenOCD-Nuvoton/issues/10#issue-696046539

Sincerely, 
Zale Yu

vincentdupaquis, 2020-09-17:
> #10 (comment)

Hello,

Thanks for being so reactive. At the end, I simply found what I was
searching for in the NuEclipse package.

Thanks anyway, I keep this in case ...

Best regards,
Vincent.

closed 2020-09-18 by ZaleYu