OpenNuvoton/OpenOCD-Nuvoton #14: Debugging NuMaker-M2351 not working

danielRep, 2021-04-29:
Hi,

I recently started to familiarize with the NuMaker-M2351 board. Since I normally prioritize open-source tools, I was glad to see that Nuvoton had a fork of OpenOCD that I could use to flash and debug the platform. However, I have been facing some issues which I hope you can help me to solve. 

I start the openocd bridge by executing the following command:
`./src/openocd -f tcl/interface/nulink.cfg -f tcl/target/numicroM23_NS.cfg`

Then I start the gdb session, connect to the openocd port, specify the binary and perform a reset
`(gdb) target remote :3333`
`(gdb) file /path/to/file.elf`
`(gdb) monitor reset init`

After I query OpenOCD to list the flash module:
`(gdb) monitor flash banks`

Ouput:
> 0 : NuMicro.flash_aprom (numicro) at 0x00000000, size 0x10080000, buswidth 0, chipwidth 0
> 1 : NuMicro.flash_data (numicro) at 0x0001f000, size 0x00000000, buswidth 0, chipwidth 0
> 2 : NuMicro.flash_ldrom (numicro) at 0x00100000, size 0x00001000, buswidth 0, chipwidth 0
> 3 : NuMicro.flash_config (numicro) at 0x00300000, size 0x00000010, buswidth 0, chipwidth 0

0 APROM: 512KB
1 actualy I don't know what region is this.
2 LDROM: 4KB
3 CONFIG-BLK: 32B

This are my binary file to be loaded section headers:
[image]

After I erase the flash: 
`(gdb) monitor numicro M2351_erase`
> numicro M2351_erase complete

Then I try to load the binary:
`(gdb) load`
> Loading section .text, size 0x10da0 lma 0x10000
> Loading section .rodata, size 0x7014 lma 0x20da0
> Loading section .ARM.exidx, size 0x8 lma 0x27db4
> Loading section .gnu.sgstubs, size 0x80 lma 0x28000
> Loading section .data, size 0x9e8 lma 0x28800
> Start address 0x102bc, load size 100388
> Transfer rate: 49017 KB/sec, 10038 bytes/write.

I place a hardware breakpoint on the address of the first instruction of the Reset_Handler and execute the command continue
`(gdb) hb *0x000102bd`

What I would expected now is that the program to be debugged from now on, but it seems that the CPU is locked at least from the ouput from gdb: 
[image]

Can you help me to figure out what is going on?

ZaleYu, 2021-04-30:
Hi,
The numicroM23_NS.cfg file is used by non-secure program. If you tried to develop a non-secure program, a secure program must be downloaded in advance. Have you prepared any secure program and download it into your M2351 target chip before developing your non-secure program?

Sincerely, 
Zale Yu

danielRep, 2021-04-30:
> Hi,
> The numicroM23_NS.cfg file is used by non-secure program. If you tried to develop a non-secure program, a secure program must be downloaded in advance. Have you prepared any secure program and download it into your M2351 target chip before developing your non-secure program?
> 
> Sincerely,
> Zale Yu

Hi Mr Yu 

I add a typo, and I am using the numicroM23.cfg file since I am trying to flash my secure program. Meanwhile I have investigate a little bit further and I did the following:

Started openocd:
`./src/openocd -f tcl/interface/nulink.cfg -f tcl/target/numicroM23.cfg`

Started GDB and executed the following commands: 
`target remote :3333`
`monitor reset init`
`monitor numicro M2351_erase`
`monitor flash write_image /path/to/file.elf 0x00010000`

Output:
[image]

However, if I try to read the flash memory, it seems it wasn't flashed:
[image]

Any idea Mr Yu?

ZaleYu, 2021-04-30:
Hi Mr. Oliveira:
Could you download the NuEclipse IDE, as follows, and read chapter 3.2 in the user manual.
https://www.nuvoton.com/support/tool-and-software/software/dirver/?__locale=zh_TW

Try to create a M2351 secure project from New Project Wizard. Then build it and download the binary to the target chip.
I want to make sure whether your environment is fine or not. Thanks. 

Sincerely, 
Zale Yu

danielRep, 2021-04-30:
Well, Mr Yu after writing the last reply I had an idea regarding the offset that I have passed during the flash command. Since I am using an elf file I think I don't need that offset, and now the code appears in the expected address: 
[image]
Previously it was flashed on 0x00020000.

Let me continue to explore this setup, and if I don't find any more problems, I will close this issue.

ZaleYu, 2021-04-30:
Hi  Mr. Oliveira:
That is great. Don't mind and keep going. 
Wish you have a good development experience with M2351 chip.

Sincerely, 
Zale Yu

danielRep, 2021-04-30:
Everything running smoothly. 
More details [here](https://github.com/Samsung/mTower/issues/31). 
Closed.

closed 2021-04-30 by danielRep