OpenNuvoton/OpenOCD-Nuvoton #20: No program output shown in gdb using Nu-Link1 or Nu-Link2 Pro

c01dkit, 2022-02-17:
Hi there,
I'm trying to trace instructions of a sample TrustZone program (https://github.com/OpenNuvoton/M2351BSP/tree/master/SampleCode/TrustZone/CSSD_LED) on NuMaker-PFM-M2351 board. I use Keil to flash the .bin file to the board, then connect the board to Windows 10, using XShell to do serial communication. Here is the screenshot:
[image]
The green LED and orange LED are flashing normally, and I can get the output of the program using XShell. Here is the connection setting. The four switches status: TXD on , RXD on, VCOM on and MSG off.
[image]

However, when I use openocd as gdb server and try to debug the program, no output produced. Here is the screenshot:
[image]

The two LEDs are still flashing normally as before, so I think the program is running normally. In my expectation, there should be some  
output such as "Secure/Non-secure LED Off call by Secure" appears in the terminal.

Futhermore, I tried Nu-Link2 Pro as the debug dongle, but it remains the same, as the following pictures:

[image]

[image]

Is the output redirected to somewhere else? If so, where is it? If not, how could I get the output?

Looking forward to help.

ccli0, 2022-02-21:
Hi c01dkit:

Using XShell to do serial communication and the output would display on the terminal.
Like the first screenshot mentioned in your post.

Sincerely,
CCLI0

c01dkit, 2022-02-21:
> Hi c01dkit:
> 
> Using XShell to do serial communication and the output would display on the terminal. Like the first screenshot mentioned in your post.
> 
> Sincerely, CCLI0

Hi ccli0,

Yeah that's what I expect to see in openocd... How could I get the same output in gdb (using openocd as gdb server)?  It seems that printf() and scanf() cannot work in that case. I've tried  enabled semihosting, but it couldn't work either.

Sincerely,
c01dkit

ccli0, 2022-02-21:
Hi c01dkit:

If you don't want to use terminal console to print messages,
please refer the following semihosting example and download code
https://github.com/OpenNuvoton/M2351BSP/tree/master/SampleCode/Semihost

start OpenOCD as gdb server and use GDB to connect to server.
[image]

The printf message would show on OpenOCD console.
[image]

c01dkit, 2022-02-21:
Hi ccli0,

Thanks for your warm help! Semihosting does work well on this simple project ( I've tried and verified), but it still got stuck when I tried semihosting on a particular project ( [mTower](https://github.com/Samsung/mTower) ). I compared the board initializing code of these two projects and found that mTower also contains the following code snippet (in [arch/cortex-m23/m2351/src/numaker_pfm_m2351/secure/main.c](https://github.com/Samsung/mTower/blob/master/arch/cortex-m23/m2351/src/numaker_pfm_m2351/secure/main.c#L115) ) :

```c
SYS->GPB_MFPH = (SYS->GPB_MFPH & (~(UART0_RXD_PB12_Msk | UART0_TXD_PB13_Msk)))
      | UART0_RXD_PB12 | UART0_TXD_PB13;
```

However, semihosting just not work when I testing mTower in the same way as before. Maybe I overlooked some details on this  particular project?

ccli0, 2022-02-21:
Hi c01dkit:

Do you have try this or other semihost discussion article from website?
https://bgamari.github.io/posts/2014-10-31-semihosting.html
Please check the steps and compare the semihost example to modify your project.

The semihosting example need to add some extra code(initialise_monitor_handles()) to main function before using printf.
If you are still stuck, it is recommended to use the COMPORT terminal to print messages.
We usually use this method.

Sincerely,
CCLI0

c01dkit, 2022-02-21:
Hi ccli0,

Really thanks for your help! I think this solution will work.

Sincerely,
c01dkit

closed 2022-02-21 by c01dkit