OpenNuvoton/OpenOCD-Nuvoton #21: How to Flash app into APROM?

KunYi, 2022-03-09:
Hi there,
        How to flash application binary into APROM by OpenOCD-Nuvoton?

I try the below commands, always failed, the OpenOCD copy from NuEclipse_V1.01.019_Windows
------------------------------------------------------------------------------------------------------
Open On-Chip Debugger 0.10.0-dev-00476-gd1de37d6-dirty (2021-12-08-10:03)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : NULINK is Nu-Link1
Info : NULINK firmware_version(7313), product_id(0x40012009)
Info : IDCODE: 0x2BA01477
Info : NuMicro.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'telnet' connection on tcp/4444
NuMicro.cpu: target state: halted
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x00000548 msp: 0x20028000
-------------------------------------------------------------------------------------------------------------------------
> reset init
> flash write_image erase unlock C:/Users/kunic/source/repos/M480_VSCODE/build/M480_Templete.bin
auto erase enabled
auto unlock enabled
Device ID: 0x00d48750
Device Name: M487JIDAE
bank base = 0x00000000, size = 0x10080000

Connection to host lost.

KunYi, 2022-03-09:
but the openocd support *write_bank* command
```
> flash banks
#0 : NuMicro.flash_aprom (numicro) at 0x00000000, size 0x00000000, buswidth 0, chipwidth 0
#1 : NuMicro.flash_data (numicro) at 0x0001f000, size 0x00000000, buswidth 0, chipwidth 0
#2 : NuMicro.flash_ldrom (numicro) at 0x00100000, size 0x00000000, buswidth 0, chipwidth 0
#3 : NuMicro.flash_config (numicro) at 0x00300000, size 0x00000000, buswidth 0, chipwidth 0
#4 : NuMicro.flash_dfmc_data (numicro) at 0x00400000, size 0x00000000, buswidth 0, chipwidth 0
> flash write_bank 0 C:/Users/kunic/source/repos/M480_VSCODE/build/M480_Templete.bin
embedded:startup.tcl:33: Error:
in procedure 'flash'
in procedure 'ocd_bouncer'
at file "embedded:startup.tcl", line 33
> flash write_bank 0 C:/Users/kunic/source/repos/M480_VSCODE/build/M480_Templete.bin 0
Device ID: 0x00d48750
Device Name: M487JIDAE
bank base = 0x00000000, size = 0x10080000
Nuvoton NuMicro: Flash Write ...
Have written 84%
Have written 100%
wrote 8676 bytes from file C:/Users/kunic/source/repos/M480_VSCODE/build/M480_Templete.bin to flash bank 0 at offset 0x00000000 in 3.714321s (2.281 KiB/s)
>
```

ccli0, 2022-03-09:
Hi KunYi,

Just use bellow command to write APROM.
flash write_image erase C:/Users/kunic/source/repos/M480_VSCODE/build/M480_Templete.bin
Thanks

KunYi, 2022-03-09:
@ccli0 , Thank you for your answer, 
Awesome! got correct response now.
```
> flash write_image erase C:/Users/kunic/source/repos/M480_VSCODE/build/M480_Template.bin
auto erase enabled
Device ID: 0x00d48750
Device Name: M487JIDAE
bank base = 0x00000000, size = 0x10080000
Nuvoton NuMicro: Sector Erase ... (0 to 2)
Nuvoton NuMicro: Flash Write ...
Have written 59%
Have written 100%
wrote 12288 bytes from file C:/Users/kunic/source/repos/M480_VSCODE/build/M480_Template.bin in 5.001838s (2.399 KiB/s)
```

closed 2022-03-09 by KunYi

KunYi, 2022-03-09:
debugging & programming with VSCode
[image]