OpenNuvoton/OpenOCD-Nuvoton #22: Problem with NuMaker-PFM-M487

Ed-Yang, 2022-03-20:
I build and run OpenOCD-Nuvoton on a Raspberry Pi 3, and failed to establish the debug session.  

Enclosed is the command and logs, please advice what wrong with my setup.  Thanks!

Besides, Nuvoton provides both OpenOCD and pyOCD, which one is recommended ?

-----------------------------------------------
$ sudo openocd -f ./tcl/interface/nulink.cfg -f ./tcl/target/numicroM4.cfg -d4

Open On-Chip Debugger 0.10.0-dev-00477-g57b40002 (2022-03-20-08:29)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 8 command.c:544 command_print(): debug_level: 4
Debug: 14 9 options.c:166 add_default_dirs(): bindir=/usr/local/bin
Debug: 15 9 options.c:167 add_default_dirs(): pkgdatadir=/usr/local/share/openocd
Debug: 16 9 options.c:168 add_default_dirs(): run_prefix=/usr/local
Debug: 17 9 configuration.c:42 add_script_search_dir(): adding /root/.openocd
Debug: 18 9 configuration.c:42 add_script_search_dir(): adding /usr/local/usr/local/share/openocd/site
Debug: 19 9 configuration.c:42 add_script_search_dir(): adding /usr/local/usr/local/share/openocd/scripts
Debug: 20 9 configuration.c:82 find_file(): found ./tcl/interface/nulink.cfg
Debug: 21 10 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_interface hla
Debug: 22 10 command.c:143 script_debug(): command - interface ocd_interface hla
Debug: 24 10 command.c:364 register_command_handler(): registering 'ocd_hla_device_desc'...
Debug: 25 10 command.c:364 register_command_handler(): registering 'ocd_hla_serial'...
Debug: 26 10 command.c:364 register_command_handler(): registering 'ocd_hla_layout'...
Debug: 27 10 command.c:364 register_command_handler(): registering 'ocd_hla_close'...
Debug: 28 10 command.c:364 register_command_handler(): registering 'ocd_hla_reset'...
Debug: 29 10 command.c:364 register_command_handler(): registering 'ocd_hla_vid_pid'...
Debug: 30 10 command.c:364 register_command_handler(): registering 'ocd_hla_vids_pids'...
Debug: 31 11 command.c:364 register_command_handler(): registering 'ocd_hla_command'...
Debug: 32 11 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla_layout nulink
Debug: 33 11 command.c:143 script_debug(): command - hla_layout ocd_hla_layout nulink
Debug: 35 11 hla_interface.c:239 hl_interface_handle_layout_command(): hl_interface_handle_layout_command
Debug: 36 11 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla_vids_pids 0x0416 0x511b 0x0416 0x511c 0x0416 0x511d
Debug: 37 11 command.c:143 script_debug(): command - hla_vids_pids ocd_hla_vids_pids 0x0416 0x511b 0x0416 0x511c 0x0416 0x511d
Debug: 39 11 hla_interface.c:312 hl_interface_handle_vids_pids_command(): hl_interface_handle_vids_pids_command
Debug: 40 11 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla_device_desc Nu-Link
Debug: 41 12 command.c:143 script_debug(): command - hla_device_desc ocd_hla_device_desc Nu-Link
Debug: 43 12 hla_interface.c:213 hl_interface_handle_device_desc_command(): hl_interface_handle_device_desc_command
Debug: 44 12 configuration.c:82 find_file(): found ./tcl/target/numicroM4.cfg
Debug: 45 12 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 46 13 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Info : 47 13 transport.c:286 jim_transport_select(): auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Debug: 48 13 hla_transport.c:191 hl_transport_select(): hl_transport_select
Debug: 49 13 command.c:364 register_command_handler(): registering 'ocd_hla'...
Debug: 50 13 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 51 13 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 52 13 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 53 13 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 54 13 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 55 13 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 56 13 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 57 14 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 58 14 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 59 14 command.c:364 register_command_handler(): registering 'ocd_jtag_ntrst_delay'...
Debug: 60 14 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 61 14 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Debug: 62 14 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 63 15 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Debug: 64 15 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 65 15 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Debug: 66 15 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla newtap NuMicro cpu -irlen 4 -expected-id 0x2BA01477
Debug: 67 15 command.c:143 script_debug(): command - ocd_hla ocd_hla newtap NuMicro cpu -irlen 4 -expected-id 0x2BA01477
Debug: 68 15 hla_tcl.c:116 jim_hl_newtap_cmd(): Creating New Tap, Chip: NuMicro, Tap: cpu, Dotted: NuMicro.cpu, 4 params
Debug: 69 15 hla_tcl.c:126 jim_hl_newtap_cmd(): Processing option: -irlen
Debug: 70 15 hla_tcl.c:126 jim_hl_newtap_cmd(): Processing option: -expected-id
Debug: 71 16 core.c:1304 jtag_tap_init(): Created Tap: NuMicro.cpu @ abs position 0, irlen 0, capture: 0x0 mask: 0x0
Debug: 72 16 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_target create NuMicro.cpu cortex_m -chain-position NuMicro.cpu
Debug: 73 16 command.c:143 script_debug(): command - ocd_target ocd_target create NuMicro.cpu cortex_m -chain-position NuMicro.cpu
[... 300 lines cut ...]
Debug: 384 700 nulink_usb.c:150 print64BytesBufferContent(): 08 04 e2 00 00 00 09 20 01 40 80 88 5f 38 37 04 
Debug: 385 700 nulink_usb.c:150 print64BytesBufferContent(): 3b 04 00 84 00 00 72 00 87 bb 00 00 bf ca 00 72 
Debug: 386 700 nulink_usb.c:150 print64BytesBufferContent(): 17 36 85 b5 c8 3b b8 f4 e4 af 08 90 53 a3 c4 34 
Debug: 387 700 nulink_usb.c:150 print64BytesBufferContent(): f1 c4 99 ee 77 a6 98 92 dc 68 66 54 fd b2 98 ce 
Debug: 388 700 nulink_usb.c:503 nulink_usb_reset(): nulink_usb_reset: RESET_HW
Debug: 389 700 nulink_usb.c:533 nulink_usb_reset(): nulink_usb_reset: EXTMODE_M030G
Debug: 390 702 nulink_usb.c:141 print64BytesBufferContent(): cmd transferred:
Debug: 391 702 nulink_usb.c:150 print64BytesBufferContent(): 09 10 e2 00 00 00 01 00 00 00 00 00 00 00 00 00 
Debug: 392 702 nulink_usb.c:150 print64BytesBufferContent(): 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Debug: 393 702 nulink_usb.c:150 print64BytesBufferContent(): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Debug: 394 702 nulink_usb.c:150 print64BytesBufferContent(): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Debug: 395 721 nulink_usb.c:141 print64BytesBufferContent(): data received:
Debug: 396 721 nulink_usb.c:150 print64BytesBufferContent(): 09 04 e2 00 00 00 09 20 01 40 80 88 5f 38 37 04 
Debug: 397 721 nulink_usb.c:150 print64BytesBufferContent(): 3b 04 00 84 00 00 72 00 87 bb 00 00 bf ca 00 72 
Debug: 398 721 nulink_usb.c:150 print64BytesBufferContent(): 17 36 85 b5 c8 3b b8 f4 e4 af 08 90 53 a3 c4 34 
Debug: 399 721 nulink_usb.c:150 print64BytesBufferContent(): f1 c4 99 ee 77 a6 98 92 dc 68 66 54 fd b2 98 ce 
Error: 400 721 nulink_usb.c:1652 nulink_usb_open(): nulink_usb_reset failed
Debug: 401 721 hla_layout.c:47 hl_layout_open(): failed
Debug: 402 721 command.c:626 run_command(): Command failed with error code -4
User : 403 722 command.c:687 command_run_line(): in procedure 'init' 
in procedure 'ocd_bouncer'
Debug: 404 722 command.c:626 run_command(): Command failed with error code -4
User : 405 722 command.c:687 command_run_line(): 
Debug: 406 722 hla_interface.c:117 hl_interface_quit(): hl_interface_quit
Debug: 407 722 nulink_usb.c:1431 nulink_usb_close(): nulink_usb_close

Ed-Yang, 2022-03-20:
After reboot RPI3 and run again, the error is gone ....

silver-kuusik, 2023-04-18:
Hi, I am also having the same issue and it seems like a system reboot solves the NuLink reset issue. I would appreciate a better solution, like resetting the USB driver or etc.

Open On-Chip Debugger 0.10.0-dev-00483-gcc9d6431-dirty (2022-09-28-08:49)
Info : clock speed 1000 kHz
Info : NULINK is Nu-Link1
Info : NULINK firmware_version(7215), product_id(0x40012009)
Info : NULINK is Nu-Link-Pro, target_voltage_mv(3371), usb_voltage_mv(5183)

@Ed-Yang did you further look into this

silver-kuusik, 2023-04-20:
@ccli0 I think this is a duplicate issue with: https://github.com/OpenNuvoton/OpenOCD-Nuvoton/issues/29 so we can close it actually. I am having the same issue as @Ed-Yang saw I think on RPI3 with arch linux. It only happens sometimes.

closed 2023-04-20 by ccli0

silver-kuusik, 2023-04-21:
> @silver-kuusik FYI-I just do some 'test' with this setup, I did not look into the problem.

I think I was seeing the same issue as you saw, only sometimes the programmer reset was failing and it seems like it does not happen with the latest programmer firmware. Just update it using the Windows ICP tool. Check here: https://github.com/OpenNuvoton/OpenOCD-Nuvoton/issues/29#issuecomment-1515879499