OpenNuvoton/OpenOCD-Nuvoton #27: The correct APROM size of Nano100BN chips

starnight, 2022-11-08:
Current OpenOCD-Nuvoton shows some [Nano100BN](https://github.com/OpenNuvoton/OpenOCD-Nuvoton/blob/b78f50ab953b2a70d733a5f4e36b420a4a2ae834/src/flash/nor/numicro.c#L421) chips have `123*1024=123KB` APROM.  
The section "5.6.2 Features" of [NuMicro® Family Nano100 Series Datasheet](https://www.nuvoton.com/export/resource-files/DS_Nano100%28B%29_Series_EN_Rev1.09.pdf) mentions `32/64/123 KB application program memory (APROM)`.

However, according to "Figure 3.1-1 NuMicro® Nano100 Series Selection Code" and "Table 3.2-1 Nano100 Base Line Selection Table" of [NuMicro® Family Nano100 Series Datasheet](https://www.nuvoton.com/export/resource-files/DS_Nano100%28B%29_Series_EN_Rev1.09.pdf), say the chips have **128 KB** Flash, instead of **123 KB**, for example NANO100KE3BN.

It will be good to understand the difference between the values.

ccma1120, 2022-11-09:
Hi, 
Nano APROM is 123KB 
128KB on document means 123KB APROM+4KB LDROM+1KB Config = 128KB

starnight, 2022-11-09:
Thanks for clarification!

closed 2022-11-09 by starnight