OpenNuvoton/OpenOCD-Nuvoton #29: Error: nulink_usb_reset failed

girishji, 2023-01-27:
I am using Numaker-etm-m487 evaluation board, and I get this.

```
> sudo openocd -f eclipse/OpenOCD/scripts/interface/nulink.cfg -f eclipse/OpenOCD/scripts/target/numicroM4.cfg
Open On-Chip Debugger 0.10.0-dev-00486-g43a1564e-dirty (2023-01-27-11:52)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : NULINK is Nu-Link1
Error: nulink_usb_reset failed
in procedure 'init'
in procedure 'ocd_bouncer'

```

smirkybg, 2023-01-27:
Is your USB Controller with versioh 3.**_0_**?

girishji, 2023-01-27:
I think not. I plugged evaluation board programmer to macbook air M1.

ccli0, 2023-01-30:
Please add -d3 option and get more information.
sudo openocd -f eclipse/OpenOCD/scripts/interface/nulink.cfg -f eclipse/OpenOCD/scripts/target/numicroM4.cfg -d3

girishji, 2023-01-30:
Thank you for reply. Here is the trace you requested.
--
 sudo openocd -f eclipse/OpenOCD/scripts/interface/nulink.cfg -f eclipse/OpenOCD/scripts/target/numicroM4.cfg -d3  2>&1|tee ocderr.out
Open On-Chip Debugger 0.10.0-dev-00486-g43a1564e-dirty (2023-01-27-11:52)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 7 command.c:544 command_print(): debug_level: 3
Debug: 14 7 options.c:172 add_default_dirs(): bindir=/usr/local/bin
Debug: 15 7 options.c:173 add_default_dirs(): pkgdatadir=/usr/local/share/openocd
Debug: 16 7 options.c:174 add_default_dirs(): run_prefix=/usr/local
Debug: 17 7 configuration.c:42 add_script_search_dir(): adding /Users/gp/.openocd
Debug: 18 7 configuration.c:42 add_script_search_dir(): adding /usr/local/usr/local/share/openocd/site
Debug: 19 7 configuration.c:42 add_script_search_dir(): adding /usr/local/usr/local/share/openocd/scripts
Debug: 20 7 configuration.c:82 find_file(): found eclipse/OpenOCD/scripts/interface/nulink.cfg
Debug: 21 7 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_interface hla
Debug: 22 7 command.c:143 script_debug(): command - interface ocd_interface hla
Debug: 24 7 command.c:364 register_command_handler(): registering 'ocd_hla_device_desc'...
Debug: 25 7 command.c:364 register_command_handler(): registering 'ocd_hla_serial'...
Debug: 26 7 command.c:364 register_command_handler(): registering 'ocd_hla_layout'...
Debug: 27 7 command.c:364 register_command_handler(): registering 'ocd_hla_close'...
Debug: 28 7 command.c:364 register_command_handler(): registering 'ocd_hla_reset'...
Debug: 29 7 command.c:364 register_command_handler(): registering 'ocd_hla_vid_pid'...
Debug: 30 7 command.c:364 register_command_handler(): registering 'ocd_hla_vids_pids'...
Debug: 31 7 command.c:364 register_command_handler(): registering 'ocd_hla_command'...
Debug: 32 7 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla_layout nulink
Debug: 33 7 command.c:143 script_debug(): command - hla_layout ocd_hla_layout nulink
Debug: 35 7 hla_interface.c:239 hl_interface_handle_layout_command(): hl_interface_handle_layout_command
Debug: 36 7 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla_vids_pids 0x0416 0x511b 0x0416 0x511c 0x0416 0x511d
Debug: 37 7 command.c:143 script_debug(): command - hla_vids_pids ocd_hla_vids_pids 0x0416 0x511b 0x0416 0x511c 0x0416 0x511d
Debug: 39 7 hla_interface.c:312 hl_interface_handle_vids_pids_command(): hl_interface_handle_vids_pids_command
Debug: 40 7 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla_device_desc Nu-Link
Debug: 41 7 command.c:143 script_debug(): command - hla_device_desc ocd_hla_device_desc Nu-Link
Debug: 43 7 hla_interface.c:213 hl_interface_handle_device_desc_command(): hl_interface_handle_device_desc_command
Debug: 44 7 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla_vids_pids 0x0416 0x511b 0x0416 0x511c 0x0416 0x511d
Debug: 45 7 command.c:143 script_debug(): command - hla_vids_pids ocd_hla_vids_pids 0x0416 0x511b 0x0416 0x511c 0x0416 0x511d
Debug: 47 7 hla_interface.c:312 hl_interface_handle_vids_pids_command(): hl_interface_handle_vids_pids_command
Debug: 48 7 configuration.c:82 find_file(): found eclipse/OpenOCD/scripts/target/numicroM4.cfg
Debug: 49 7 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 50 7 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Info : 51 8 transport.c:286 jim_transport_select(): auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Debug: 52 8 hla_transport.c:191 hl_transport_select(): hl_transport_select
Debug: 53 8 command.c:364 register_command_handler(): registering 'ocd_hla'...
Debug: 54 8 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 55 8 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 56 8 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 57 8 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 58 8 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 59 8 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 60 8 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 61 8 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 62 8 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 63 8 command.c:364 register_command_handler(): registering 'ocd_jtag_ntrst_delay'...
Debug: 64 8 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 65 8 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Debug: 66 8 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 67 8 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Debug: 68 8 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 69 8 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Debug: 70 8 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla newtap NuMicro cpu -irlen 4 -expected-id 0x2BA01477
[... cut ...]
Error: 284 17 nulink_usb.c:1650 nulink_usb_open(): open failed
Debug: 285 17 hla_layout.c:47 hl_layout_open(): failed
Debug: 286 17 command.c:626 run_command(): Command failed with error code -4
User : 287 17 command.c:687 command_run_line(): in procedure 'init'
in procedure 'ocd_bouncer'
Debug: 288 17 command.c:626 run_command(): Command failed with error code -4
User : 289 17 command.c:687 command_run_line():
Debug: 290 17 hla_interface.c:117 hl_interface_quit(): hl_interface_quit
Debug: 291 17 nulink_usb.c:1493 nulink_usb_close(): nulink_usb_close

ccli0, 2023-01-31:
Hi girishji,

According to the log, the HID device cannot be successfully enabled on your system
Please confirm whether the device PID has access rights on your MACBOOK.

silver-kuusik, 2023-04-18:
I am having the same issue, it seems like sometimes it only fails this way and then a kernel reboot helps and it works again. @girishji did you find out more?

girishji, 2023-04-18:
It is to do with native MacOS USB driver IIRC. There are some suggestions on OpenOCD page to replace the driver with open source one, but I did not pursue that road.

silver-kuusik, 2023-04-19:
> It is to do with native MacOS USB driver IIRC. There are some suggestions on OpenOCD page to replace the driver with open source one, but I did not pursue that road.

Ok, thanks for the reply. Did you have the issue each time you tried to program or sometimes. I am using archlinux on a RPI3 and seeing this issue only very rarely. We use OpenOCD and the NuLink Pro in a 24/7 CI / CD.

girishji, 2023-04-19:
I had issue each time.

ccli0, 2023-04-20:
I am not familiar with MacOS.
Does MacOS have the control setting to manage USB access permissions like ubuntu?

In ubuntu
When accessing target chips via Nu-Link, GNU/Linux requires the USB permission. We can get the permission by adding udev rules for Nu-Link.
Add the User to the plugdev Group. Type the command in the Terminal:
sudo useradd -G plugdev $USER
– Add Nu-Link to udev. Type the commands in the Terminal:
cd /etc/udev/rules.d and sudo gedit 10-openocd-nulink.rules
– Add the following text to the file

ATTRS{idProduct}=="511b", ATTRS{idVendor}=="0416", MODE="0666", GROUP="plugdev"
ATTRS{idProduct}=="511c", ATTRS{idVendor}=="0416", MODE="0666", GROUP="plugdev"
ATTRS{idProduct}=="511d", ATTRS{idVendor}=="0416", MODE="0666", GROUP="plugdev"
ATTRS{idProduct}=="5200", ATTRS{idVendor}=="0416", MODE="0666", GROUP="plugdev"
ATTRS{idProduct}=="5201", ATTRS{idVendor}=="0416", MODE="0666", GROUP="plugdev"
ATTRS{idProduct}=="5202", ATTRS{idVendor}=="0416", MODE="0666", GROUP="plugdev"
ATTRS{idProduct}=="5203", ATTRS{idVendor}=="0416", MODE="0666", GROUP="plugdev"
ATTRS{idProduct}=="5204", ATTRS{idVendor}=="0416", MODE="0666", GROUP="plugdev"
ATTRS{idProduct}=="5205", ATTRS{idVendor}=="0416", MODE="0666", GROUP="plugdev"
ATTRS{idProduct}=="2004", ATTRS{idVendor}=="0416", MODE="0666", GROUP="plugdev"
ATTRS{idProduct}=="2005", ATTRS{idVendor}=="0416", MODE="0666", GROUP="plugdev"
ATTRS{idProduct}=="2006", ATTRS{idVendor}=="0416", MODE="0666", GROUP="plugdev"
ATTRS{idProduct}=="2007", ATTRS{idVendor}=="0416", MODE="0666", GROUP="plugdev"
ATTRS{idProduct}=="2008", ATTRS{idVendor}=="0416", MODE="0666", GROUP="plugdev"

girishji, 2023-04-20:
I think plugdev and permissions are specific to Ubuntu. MacOS does not have it. macOS drivers are closed source I think. Better to work with Linux only.

silver-kuusik, 2023-04-20:
For me the issue came up only occasionally and it seems for now it has disappeared completely with the newest software:

We have a 24 / 7 CI / CD setup:
- RPI 3 with archlinux: 5.15.76-2-rpi-ARCH 
- NuLink Pro FW 7380
- OpenOCD 0.10.022.0-dev-00487-g6088e957

I will report once I see this issue again.

silver-kuusik, 2023-05-08:
@ccli0 I managed to reproduce the issue after 3 weeks continuous running of the NuLink Pro in the CI / CD. This is the error log, would be curious if it tells more what is the issue?

```bash
openocd -d 3 -f /etc/openocd.cfg -c init -c reset -c exit
Open On-Chip Debugger 0.10.022.0-dev-00487-g6088e957-dirty (2023-04-18-11:24)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 8 command.c:544 command_print(): debug_level: 3
Debug: 14 8 options.c:166 add_default_dirs(): bindir=/usr/local/bin
Debug: 15 8 options.c:167 add_default_dirs(): pkgdatadir=/usr/local/share/openocd
Debug: 16 8 options.c:168 add_default_dirs(): run_prefix=/usr/local
Debug: 17 8 configuration.c:42 add_script_search_dir(): adding /home/alarm/.openocd
Debug: 18 8 configuration.c:42 add_script_search_dir(): adding /usr/local/usr/local/share/openocd/site
Debug: 19 8 configuration.c:42 add_script_search_dir(): adding /usr/local/usr/local/share/openocd/scripts
Debug: 20 8 configuration.c:82 find_file(): found /etc/openocd.cfg
Debug: 21 9 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_interface hla
Debug: 22 9 command.c:143 script_debug(): command - interface ocd_interface hla
Debug: 24 9 command.c:364 register_command_handler(): registering 'ocd_hla_device_desc'...
Debug: 25 9 command.c:364 register_command_handler(): registering 'ocd_hla_serial'...
Debug: 26 9 command.c:364 register_command_handler(): registering 'ocd_hla_layout'...
Debug: 27 9 command.c:364 register_command_handler(): registering 'ocd_hla_close'...
Debug: 28 9 command.c:364 register_command_handler(): registering 'ocd_hla_reset'...
Debug: 29 9 command.c:364 register_command_handler(): registering 'ocd_hla_vid_pid'...
Debug: 30 9 command.c:364 register_command_handler(): registering 'ocd_hla_vids_pids'...
Debug: 31 9 command.c:364 register_command_handler(): registering 'ocd_hla_command'...
Debug: 32 9 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla_layout nulink
Debug: 33 9 command.c:143 script_debug(): command - hla_layout ocd_hla_layout nulink
Debug: 35 9 hla_interface.c:239 hl_interface_handle_layout_command(): hl_interface_handle_layout_command
Debug: 36 10 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla_device_desc Nu-Link
Debug: 37 10 command.c:143 script_debug(): command - hla_device_desc ocd_hla_device_desc Nu-Link
Debug: 39 10 hla_interface.c:213 hl_interface_handle_device_desc_command(): hl_interface_handle_device_desc_command
Debug: 40 10 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla_vids_pids 0x0416 0x511b 0x0416 0x511c 0x0416 0x511d
Debug: 41 10 command.c:143 script_debug(): command - hla_vids_pids ocd_hla_vids_pids 0x0416 0x511b 0x0416 0x511c 0x0416 0x511d
Debug: 43 10 hla_interface.c:312 hl_interface_handle_vids_pids_command(): hl_interface_handle_vids_pids_command
Debug: 44 10 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 45 10 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Info : 46 10 transport.c:285 jim_transport_select(): auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Debug: 47 10 hla_transport.c:191 hl_transport_select(): hl_transport_select
Debug: 48 10 command.c:364 register_command_handler(): registering 'ocd_hla'...
Debug: 49 10 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 50 10 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 51 10 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 52 10 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 53 10 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 54 10 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 55 10 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 56 11 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 57 11 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 58 11 command.c:364 register_command_handler(): registering 'ocd_jtag_ntrst_delay'...
Debug: 59 11 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 60 11 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Debug: 61 11 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 62 11 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Debug: 63 11 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 64 11 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Debug: 65 12 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla newtap NuMicro cpu -irlen 4 -expected-id 0x2BA01477
Debug: 66 12 command.c:143 script_debug(): command - ocd_hla ocd_hla newtap NuMicro cpu -irlen 4 -expected-id 0x2BA01477
Debug: 67 12 hla_tcl.c:115 jim_hl_newtap_cmd(): Creating New Tap, Chip: NuMicro, Tap: cpu, Dotted: NuMicro.cpu, 4 params
Debug: 68 12 hla_tcl.c:126 jim_hl_newtap_cmd(): Processing option: -irlen
Debug: 69 12 hla_tcl.c:126 jim_hl_newtap_cmd(): Processing option: -expected-id
Debug: 70 12 core.c:1300 jtag_tap_init(): Created Tap: NuMicro.cpu @ abs position 0, irlen 0, capture: 0x0 mask: 0x0
[... cut ...]
Debug: 298 914 hla_layout.c:47 hl_layout_open(): failed
Debug: 299 914 command.c:626 run_command(): Command failed with error code -4
User : 300 915 command.c:687 command_run_line(): in procedure 'init' 
in procedure 'ocd_bouncer'
Debug: 301 915 command.c:626 run_command(): Command failed with error code -4
User : 302 915 command.c:687 command_run_line(): 
Debug: 303 915 hla_interface.c:117 hl_interface_quit(): hl_interface_quit
Debug: 304 915 nulink_usb.c:1493 nulink_usb_close(): nulink_usb_close
```

ccli0, 2023-05-09:
@silver-kuusik,

In your log, the command cannot get the Nu-Link firmware version successfully.
Is the nu-link frimware version you are using too old?
Or is there a problem with the hardware itself?

In normal case,
NuLink firmware  version: 7470

Debug: 277 1620 nulink_usb.c:1527 nulink_usb_open(): nulink_usb_open
Debug: 278 1620 nulink_usb.c:1534 nulink_usb_open(): Stat Case 1: 0
Debug: 279 1805 nulink_usb.c:1538 nulink_usb_open(): Run NuLink.exe on Win32 (result: 0)
Debug: 280 2384 nulink_usb.c:1554 nulink_usb_open(): Wait NuLink.exe (result: 0)
Debug: 281 2437 libusb1_common.c:35 jtag_libusb_match(): jtag_libusb_match: vid(0x0416), pid(0x511c)
Info : 282 2437 nulink_usb.c:1638 nulink_usb_open(): NULINK is Nu-Link1
Debug: 283 2437 nulink_usb.c:1640 nulink_usb_open(): interface number: 0
Debug: 284 2453 nulink_usb.c:1641 nulink_usb_open(): rx endpoint: 0x81
Debug: 285 2453 nulink_usb.c:1642 nulink_usb_open(): tx endpoint: 0x02
Debug: 286 2453 nulink_usb.c:1643 nulink_usb_open(): max_packet_size: 64
Debug: 287 2453 nulink_usb.c:1644 nulink_usb_open(): jtag_libusb_open succeeded
Debug: 288 2469 nulink_usb.c:1650 nulink_usb_open(): detach kernel driver failed(-12)
Debug: 289 2469 nulink_usb.c:1663 nulink_usb_open(): jtag_libusb_claim_interface succeeded
Debug: 290 2469 nulink_usb.c:308 nulink_usb_version(): nulink_usb_version
Info : 291 2484 nulink_usb.c:323 nulink_usb_version(): NULINK firmware_version(7470), product_id(0x00012009)
Info : 292 2484 nulink_usb.c:331 nulink_usb_version(): NULINK is Nu-Link-Pro, target_voltage_mv(3297), usb_voltage_mv(4872)
Debug: 293 2506 nulink_usb.c:1428 nulink_speed(): nulink_speed: query(0)
Debug: 294 2506 nulink_usb.c:1447 nulink_speed(): NULINK nulink_speed: 1000
Debug: 295 2569 nulink_usb.c:1473 nulink_speed(): nulink_speed: h->hardwareConfig(1)
Info : 296 2569 nulink_usb.c:1475 nulink_speed(): NULINK target_voltage_mv[0](2590583), target_voltage_mv[1](0259), target_voltage_mv[2](77890101), if_target_power_supplied(1)

silver-kuusik, 2023-05-09:
@ccli0 Thank you for the quick response. I was running `NuLink Pro FW 7380`, now just flashed the latest `NuLink Pro FW 7470`.

The programmer is connected to the linux system, I see it with lsusb: `Bus 001 Device 007: ID 0416:511c Winbond Electronics Corp. Nuvoton Nu-Link1 ICE`

But it seems like unable to communicate with it. It only starts working after a system reboot. I am using a Raspberry Pi 3+

ccli0, 2023-05-09:
@silver-kuusik 
It only starts working after a system reboot.
-> Do you mean that openocd can read the firmware version after the system reboot?
Suppose in this case, What the operation steps can generate the above log?

silver-kuusik, 2023-05-10:
@ccli0 yes I mean that openocd works as normal again when I reboot the archlinux on the RPI 3. Meaning I can use openocd to flash again.

Reproducing this is issue is as of now very complicated, it took me 3 weeks to see this issue. We are running a 24/7 continuous integration / development where the RPI3 is flashing via the Nulink Pro with OpenOCD a M481 Nuvoton MCU. During this 3 weeks, it probably flashed the Nuvoton MCU ca 750 times with different firmwares.

I think for now it makes sense to observe if this happens again with the latest NuLinkPro firmware 7470. This might take yet another 3 weeks possibly to know.

ccli0, 2023-05-10:
@silver-kuusik
I have a few questions.
OpenOCD flashed the Nuvoton MCU ca 750 times with different firmwares during this 3 weeks. 
Why does it take 3 weeks to flash 750 times?
Flashing a firmware should not take a lot of time, or did you do other operations? (not just flashing code)

When OpenOCD connection failed, could you test whether your RPI3 USB port still working? For example, connecting other devices to test whether the RPI3 port is working normal.
Another way, connect Nu-Link pro and M481 MCU to other PC devices to check if the MCU connection is normal.

Because we don't have an RPI3 environment to test, if you can test it on a PC.

silver-kuusik, 2023-05-11:
@ccli0 It's an automated test environment. We schedules nightly tests that we are running on our system (which uses the Nuvoton M481). During day time we run the automated test environment for every commit to the our codebase.

So the MCU flashing during day time is at random times and on a set time at night.

Also another thing I noted is that with OpenOCD the flashing takes 2 minutes, compared to the Windows ICP tool, ca 8 seconds. Can that be improved somehow ?

I can do more tests when this issue occurs again, it will take it's time again. But thanks for the good tips, I will try using another device on the same port.

I will give you more input once I see the issue again.

silver-kuusik, 2023-06-22:
@ccli0 I haven't seen this issue anymore blocking us so could close it in my opinion. I had added a power cycle function also with [uhubctl](https://github.com/mvp/uhubctl) program.

ccli0, 2023-06-26:
@silver-kuusik
Thank you for your reply.
ICP tool will use the maximum usable buffer for storing programming data according to the actual SRAM size of the target chip, so it could reduce the number of times of transferring data to SRAM for reducing programming time.
OpenOCD did not get the target chip SRAM size information.
ICP tool using BULK transfer will be faster than OpenOCD using HID transfer.
We are trying to improve this problem.
CMSIS-DAP transmission may be a way we try in the future.

silver-kuusik, 2023-06-26:
@ccli0 

thank you for this answer, would be great to see the programming speed improve with OpenOCD

Do you know if there would be any other solution for the time being to improve programming speed under Linux ?

ccli0, 2023-06-26:
@silver-kuusik

Maybe you could try this solution.
Use OpenOCD commands to increase available SRAM size.
For example, resize available SRAM size to 128KB (please base on your target chip)
NuMicro.cpu configure -work-area-size 0x20000

silver-kuusik, 2023-06-27:
@ccli0

I tested this, but it just transferred the image in bigger blocks, but didn't speed up the transfer.

So it seems like the bottle neck is the HID transfer ?

ccli0, 2023-06-27:
@silver-kuusik 

In my test, I change work-area-size 0x4000 and 0x20000.
The speed has indeed increased.

OpenOCD command
openocd.exe -f ..\scripts\interface\nulink.cfg -f ..\scripts\target\numicroM4.cfg -d3

Maybe you can show your test log.

silver-kuusik, 2023-06-27:
@ccli0 

I see a slight decrease in time: 
- **0x4000**: wrote 290816 bytes from file build-lynx-dvt2/bin/scb.bin in 109.128731s (2.602 KiB/s)
- **0x20000**: wrote 290816 bytes from file build-lynx-dvt2/bin/scb.bin in 99.258873s (2.861 KiB/s)

Full log as a file from when using **0x20000** https://file.io/v5gYhT6F6r9o

ccli0, 2023-06-28:
@silver-kuusik 

I cannot get the log file from website you provided.
The transfer you requested has been deleted.

silver-kuusik, 2023-06-28:
@ccli0 

Sorry, uploaded here now: https://easyupload.io/13oc0g

ccli0, 2023-06-29:
@silver-kuusik

There seems to be nothing wrong with your log.
Maybe the bottle neck is the HID transfer.
I cannot think any other way to increase the speed unless modified source code.

silver-kuusik, 2023-06-29:
@ccli0 

Thanks a lot still, It seems to also be USB hardware or Linux kernel specific, on another machine it is even faster, 30% faster:

wrote 290816 bytes from file mega.bin in 75.711006s (3.751 KiB/s)

sk333malik, 2024-10-21:
Connect the TMS or VCC wire too for NU-LINK-PRO

multiplemonomials, 2025-04-28:
I just ran into this same issue (`nulink_usb_version failed with cmdSize(4 * 6)`) with a NUMAKER-PFM-M487 board. To add my data points:
- Upgrading firmware to the latest (from https://github.com/OpenNuvoton/Nuvoton_Tools/blob/master/Latest_NuLink_Firmware/NuLink1FW.bin) had no effect
- Using the board from Windows instead of Linux made it work fine
- Running OpenOCD with sudo had no effect
- Rebooting the board had no effect
- Rebooting the host machine fixed it, it is now working correctly!