OpenNuvoton/OpenOCD-Nuvoton #30: Request for pre-built binaries

danchouzhou, 2023-02-17:
Hello ccli0, 

I'm building vscode environment for Nuvoton. Would you please release the pre-built binaries for both Windows and Linux, so that user can download and use it directly. Thank you.

FYI:
[Nuvoton-VScode-template](https://github.com/danchouzhou/Nuvoton-VScode-template)

Cheers, 
Danny

ccma1120, 2023-02-17:
Hi Danny, 
Thanks for your feedback, we will check it soon.

ccli0, 2023-02-17:
Hi Danny,

I have no idea what you need.
I follow your instruction and build HelloWorld.bin successfully on Windows.

danchouzhou, 2023-02-17:
Hello cclin0, 

I need the OpenOCD-Nuvoton execution files, so that I can integrate it into the VScode.

Cheers, 
Danny

danchouzhou, 2023-02-20:
Any news? I know we can get it by installing NuEclipse, but it's not the good way for our users.

ccli0, 2023-02-20:
Hi Danny,

How do I give the files to you?
The file type will be two zip files. (Windows and Linux)

danchouzhou, 2023-02-20:
Hello cclin0, 

Thanks for replying! Use the ``release`` in your repository if it is possible? So that I can tell the users where to download it.
https://github.com/OpenNuvoton/OpenOCD-Nuvoton/releases

Cheers, 
Danny

ccli0, 2023-02-20:
Hi Danny,

I have uploaded the zip files.
https://github.com/OpenNuvoton/OpenOCD-Nuvoton/releases

danchouzhou, 2023-02-22:
Hello ccli0, 

Success! Check this out.
https://github.com/danchouzhou/Nuvoton-VScode-template

Thanks for helping.
Danny

closed 2023-02-23 by danchouzhou