OpenNuvoton/OpenOCD-Nuvoton #32: Possible GPL Violation openocd_8051_1T.exe

paulsykes, 2023-06-24:
I'm attempting to use openocd on a Nuvoton 8051 MCU in Linux.

The NuEclipse IDE for Windows ships with openocd.exe and openocd_8051_1T.exe .  I assumed this repo would contain the code for both openocd.exe and openocd_8051_1T.exe.  My investigation shows that it only contains code for the former, please correct me if I'm wrong?

Looking through the closed issues I found this https://github.com/OpenNuvoton/OpenOCD-Nuvoton/issues/23 the response seems incorrect. Nuvoton does ship a version of OpenOCD that supports 8051 in the NuEclipse IDE for Windows available here https://www.nuvoton.com/resource-download.jsp?tp_GUID=SW192023032902444727&currentFolder=/products/microcontrollers/8bit-8051-mcus/industrial-8051-series/

ccli0, 2023-06-26:
@paulsykes,
Thank you for your reply.
Yes, you are right.
We will create OpenOCD repo for Nuvoton 8051.
For #23 reply, at the time we really hadn't launched support for OpenOCD 8051.

paulsykes, 2023-06-26:
@ccli0 thank you very much for your quick response. I'd like to apologise, I'd looked through the OpenNuvton repos and did not find https://github.com/OpenNuvoton/OpenOCD-Nuvoton-8051 thank you for pointing this out.

Having done a more thorough search of your repos, I'm also really happy to see https://github.com/OpenNuvoton/NuEclipse_8051_SDCC

I've recently inherited a project using an 8051 Nuvoton MCU, I tried out the NuEclipse IDE in Windows and everything was working great. Where possible I prefer to use Linux and this now looks possible.

Many thanks.

closed 2023-06-26 by paulsykes