OpenNuvoton/OpenOCD-Nuvoton #34: Any chance we will see this updated to upstream OpenOCD?

tinic, 2023-09-30:
The upstream version has no support for the M460 series, so I am stuck with OpenOCD-Nuvoton.

At the same time the current version of the Cortex-Debug plugin for VSCode is not compatible with this version of OpenOCD-Nuvoton anymore which means debugging is now impossible with your fork.

The longer you wait dealing with admittedly large task, the worse it will get. Just sayin...

ccma1120, 2023-09-30:
Thank you for your advise. We plan to transition to the CMSIS-DAP standard in the future. Currently, we are experimenting with upstreaming M4 "https://github.com/OpenNuvoton/OpenOCD-Nuvoton-CMSIS-DAP", and it is awaiting review.

WebDust21, 2023-10-14:
Is the CMSIS-DAP standard going to be in the actual OpenOCD tool, and not in a fork?

because I notice that Nuvoton also has a pyOCD fork--and it is not up to date with the real pyOCD as well.
This is a big problem for long-term maintainability.

WebDust21, 2023-10-14:
even with the proper ARM Pack for the Nuvoton processor (in my case M252KG6AE), pyOCD can not recognize either the Nu-Link-Pro or Nu-Link2-Me official Nuvoton programmers.

Neither can upstream OpenOCD (despite OpenOCD mentioning "Nu-Link" in the list of "FLASH drivers").

ccma1120, 2023-10-17:
> even with the proper ARM Pack for the Nuvoton processor (in my case M252KG6AE), pyOCD can not recognize either the Nu-Link-Pro or Nu-Link2-Me official Nuvoton programmers.
> 
> Neither can upstream OpenOCD (despite OpenOCD mentioning "Nu-Link" in the list of "FLASH drivers").

pyOCD should be able to support Nu-Link2-Me + M252KG6AE.

WebDust21, 2023-10-17:
Can you provide an example that show pyOCD recognizing Nu-Link2-Me programmer?

I have tried the following:
`pyocd list --pack="/mnt/Files/Nuvoton/Nuvoton.NuMicro_DFP.1.3.21.pack"`

And always get the following response:
`No available debug probes are connected`

ccma1120, 2023-10-18:
> Can you provide an example that show pyOCD recognizing Nu-Link2-Me programmer?
> 
> I have tried the following: `pyocd list --pack="/mnt/Files/Nuvoton/Nuvoton.NuMicro_DFP.1.3.21.pack"`
> 
> And always get the following response: `No available debug probes are connected`

There should be a disk pop up when you plug in Nu-Link2-Me, and there is a file NU_CFG.txt, can you open and check the version >= 7537r, CMSIS-DAP=1  ?

WebDust21, 2023-10-18:
Version=7313r
CMSIS-DAP=0

Tried setting CMSIS-DAP to 1, but upon restart of the programmer, it was back to 0.

(I am on Linux, so cannot use the Windows tools.  And that is why OpenOCD / pyOCD are such valuable tools for me.)

ccma1120, 2023-10-19:
> Version=7313r CMSIS-DAP=0
> 
> Tried setting CMSIS-DAP to 1, but upon restart of the programmer, it was back to 0.
> 
> (I am on Linux, so cannot use the Windows tools. And that is why OpenOCD / pyOCD are such valuable tools for me.)

Is it possible to find a Windows system ? Download the latest NuMicro_ICP_Programming_Tool_V3.14. https://www.nuvoton.com/tool-and-software/software-tool/programmer-tool/ 
Install and press connect, it will prompt to upgrade the firmware to the latest 7565r, and the problem will be solved.

WebDust21, 2023-10-19:
> Is it possible to find a Windows system ?

Not easy for me.  That is why I am trying to use tools that run on Linux.