OpenNuvoton/OpenOCD-Nuvoton #35: Release "OpenOCD_V1.02.023r" not ready-to-use

danchouzhou, 2023-09-30:
Hello ccli0, 

I saw your new release, but the files comes up with only openocd.exe without script, .dll, etc. 
So the people who download the OpenOCD_Windows.zip or OpenOCD_Linux.zip are not able to use it directly.

Please take a look how I use your build. If the file an directory structure are all same for every release (such like arm-gcc), it will be great!
[Nuvoton-VScode-template](https://github.com/danchouzhou/Nuvoton-VScode-template)

：(
[OpenOCD_V1.02.023r](https://github.com/OpenNuvoton/OpenOCD-Nuvoton/releases/tag/v1.02.023r)

：)
[OpenOCD_V1.02.022r](https://github.com/OpenNuvoton/OpenOCD-Nuvoton/releases/tag/v1.02.022r)

Cheers, 
Danny

ccli0, 2023-10-02:
Hi @danchouzhou 

I have re-uploaded v1.02.023 and new version v1.02.024.
Please take a look.

danchouzhou, 2023-10-02:
Thanks for your help!

closed 2023-10-02 by danchouzhou