OpenNuvoton/OpenOCD-Nuvoton #37: Nuvoton OpenOCD and the N76E003-controller not working on Ubuntu 22.04.4

jz13138, 2024-04-04:
Hi Everyone,
I face some problems with Nuvoton OpenOCD and the N76E003-controller on `Ubuntu 22.04.4.`

The BSY Led on the Nu-Link-debugger lights up shortly.
The debugging terminates immediately without further notification.
The Console looks like this:
```
<terminated>...GDB Nuvoton Nu-Link Debugging] openocd_8051_1T:
Open On-Chip Debugger 0.10.0-dev (2023-07-06-07:19)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1 kHz
Info : NULINK is Nu-Link1
Info : NULINK firmware_version(7623), product_id(0x30012009)
Info : IDCODE: 0x00DA3650
Info : IDCODE: 0x00DA3650
Started by GNU ARM Eclipse
```

openocd_8051_1T does not show any problems:

```
${openocd_nulink_path}/openocd_8051_1T -f ../scripts/interface/nulink.cfg -f ../scripts/target/numicro8051_1T.cfg -c "init; reset halt"
Open On-Chip Debugger 0.10.0-dev (2023-07-06-07:19)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1 kHz
Info : NULINK is Nu-Link1
Info : NULINK firmware_version(7623), product_id(0x30012009)
Info : IDCODE: 0x00DA3650
Info : IDCODE: 0x00DA3650
NuMicro.cpu: target state: halted
target halted due to debug-request, pc: 0x00000000
```

The long version with -d 3 is attached.
[OpenOcd_d3_out.txt](https://github.com/OpenNuvoton/OpenOCD-Nuvoton/files/14875598/OpenOcd_d3.txt)

The same setup seems to work on Windows.

Help would be appreciated to get the linux-version running.

Kind regards,
jz13138

ccli0, 2024-04-08:
Hi @jz13138,

Please download the zip file.
https://github.com/ccli0/temp_share

Put the contents of the compressed file into the Toolchain/SDCC/gdb folder and replace the original file.
Try again.
[image]

jz13138, 2024-04-08:
Hi @ccli0 ,
the updated file solved the issue.

Thanks

closed 2024-05-31 by ccli0

xdien, 2024-06-24:
I had the same problem in linux. I think openocd sent a piece of malformed data to gdb which caused it to shut down without any notification.
`
Info : 7756 234844 numicro.c:3863 numicro_get_cpu_type(): Device Name: NUC029LAN
Info : 7757 234844 numicro.c:3907 numicro_get_flash_size(): bank base = 0x00300000, size = 0x00000004
Debug: 7758 234844 numicro.c:3996 numicro_probe(): Nuvoton pageSize: 0x200
Debug: 7759 234844 numicro.c:4079 numicro_probe(): target name: common
Debug: 7760 234844 numicro.c:4104 numicro_probe(): Nuvoton NuMicro: Probed ...
Debug: 7761 234844 gdb_server.c:989 gdb_new_connection(): New GDB Connection: 1, Target NuMicro.cpu, state: running
Debug: 7762 234844 gdb_server.c:2716 gdb_input_inner(): received packet: 'qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+;memory-tagging+;xmlRegisters=i386'
Debug: 7763 234844 gdb_server.c:2716 gdb_input_inner(): received packet: 'vCont?'
Debug: 7764 234844 gdb_server.c:2716 gdb_input_inner(): received packet: 'vMustReplyEmpty'
Debug: 7765 234844 gdb_server.c:2716 gdb_input_inner(): received packet: 'QStartNoAckMode'
Debug: 7766 234844 gdb_server.c:630 gdb_get_packet_inner(): Received first acknowledgment after entering noack mode. Ignoring it.
Debug: 7767 234844 gdb_server.c:2716 gdb_input_inner(): received packet: '!'
Debug: 7768 234844 gdb_server.c:2716 gdb_input_inner(): received packet: 'Hg0'
Debug: 7769 234844 gdb_server.c:2716 gdb_input_inner(): received packet: 'qXfer:features:read:target.xml:0,1000'
Debug: 7770 234844 gdb_server.c:2716 gdb_input_inner(): received packet: 'qTStatus'
Debug: 7771 234844 gdb_server.c:2716 gdb_input_inner(): received packet: '?'
Debug: 7772 234844 gdb_server.c:2716 gdb_input_inner(): received packet: 'qfThreadInfo'
Debug: 7773 234844 gdb_server.c:2716 gdb_input_inner(): received packet: 'Hc-1'
Debug: 7774 234844 gdb_server.c:2716 gdb_input_inner(): received packet: 'qC'
Debug: 7775 234844 gdb_server.c:2716 gdb_input_inner(): received packet: 'qAttached'
Debug: 7776 234844 gdb_server.c:2716 gdb_input_inner(): received packet: 'g'
Debug: 7777 234845 gdb_server.c:1015 gdb_connection_closed(): GDB Close, Target: NuMicro.cpu, state: running, gdb_actual_connections=0
Debug: 7778 234845 target.c:1509 target_call_event_callbacks(): target event 6 (gdb-end)
Debug: 7779 234845 target.c:1509 target_call_event_callbacks(): target event 24 (gdb-detach)
Debug: 7780 234845 target.c:4286 target_handle_event(): target: (0) NuMicro.cpu (hla_target) event: 24 (gdb-detach) action:  exitDebugging 
Debug: 7781 234845 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla_close
Debug: 7782 234845 command.c:143 script_debug(): command - hla_close ocd_hla_close

`

ccli0, 2024-06-25:
@xdien 
The target chip you are using is not an 8051 product; it is an NUC029 M0 product. Please use arm-none-eabi-gdb (default setting) instead of the gdb from SDCC folder.

xdien, 2024-06-26:
> @xdien The target chip you are using is not an 8051 product; it is an NUC029 M0 product. Please use arm-none-eabi-gdb (default setting) instead of the gdb from SDCC folder.

Thank you. I'm a bit confused between gdb for x86 and arm.   I rebuilt gdb(14.3) for arm and it connected successfully